adafruit/Adafruit-RGB-LCD-Shield-Library #7: Port to particle.io

jrubinstein, 2015-07-01:
Ada-peeps,
What's the chances of getting this ported to Particle (formerly spark)? I gave it a shot, but kind of failed :( 

Thanks,
James

ladyada, 2015-07-01:
@zsup any suggestions? most libraries 'ought to just work' if Particle is Arduino compatible

jrubinstein, 2015-07-02:
Apparently it's a little more involved
http://community.particle.io/t/how-to-include-a-library-because-im-failing/12931/3

If you're porting libraries I could also use the dht lib to be ported :D
someone has already done it but it wasn't working as of this afternoon

zsup, 2015-07-02:
At a quick glance, it looks like this should be a really easy port; it doesn't use any low-level AVR stuff, and I think the only changes that would be necessary might be wrapping some of the `#includes` in `#ifdef`s. If you're looking for help working through a port, definitely go create a thread on our forums (community.particle.io), there are a few people in the community who have often ported libraries over based on requests from others in the community!

ladyada, 2015-07-02:
thx @zsup - @jrubinstein if you submit a pull req with the tweaks we can merge :)

closed 2015-07-02 by ladyada

jrubinstein, 2015-07-03:
@mumblepins has already ported!  https://github.com/mumblepins/Adafruit_RGBLCDShield