adafruit/Adafruit-RGB-LCD-Shield-Library #9: Not sure if it works with product/1115

scls19fr, 2017-05-27:
Hello,

I bought https://www.adafruit.com/product/1115
LCD Shield Kit w/ 16x2 Character Display - Only 2 pins used! - BLUE AND WHITE

It's not very clear if code example in this project can be use to test this LCD/keypad shield.

I tried to compile and run [example](https://github.com/adafruit/Adafruit-RGB-LCD-Shield-Library/blob/master/examples/HelloWorld/HelloWorld.ino).

At startup, backlight is on but nothing appears on screen.

Any idea?

ladyada, 2017-05-29:
hiya for tech support please post in the forums
https://forums.adafruit.com
thanks!

closed 2017-05-29 by ladyada

scls19fr, 2017-05-30:
Ok sorry.
Here is forum thread https://forums.adafruit.com/viewtopic.php?f=47&t=118064