adafruit/Adafruit-RGB-LCD-Shield-Library #20: Library won't compile on Uno Wifi Rev 2

n-gineer, 2020-06-23:
- Arduino board:  **Uno Wifi Rev 2**

- Arduino IDE version (found in Arduino -> About Arduino menu):  **1.8.13**

- List the steps to reproduce the problem below (if possible attach a sketch or
  copy the sketch code in too): **Open Example sketch "Hello World"**

In trying to simply compile the Adafruit RGB LCD Shield "Hello World" example, I am getting this error:
```
Arduino: 1.8.12 (Windows Store 1.8.33.0) (Windows 10), Board: "Arduino Uno WiFi Rev2, ATMEGA328"

In file included from C:\Users\...\Arduino\libraries\Adafruit_RGB_LCD_Shield_Library\Adafruit_RGBLCDShield.cpp:17:0:

C:\Users\...\Arduino\libraries\Adafruit_RGB_LCD_Shield_Library\Adafruit_RGBLCDShield.h:68:44: error: expected class-name before '{' token

 class Adafruit_RGBLCDShield : public Print {

                                            ^

exit status 1
Error compiling for board Arduino Uno WiFi Rev2.
```

I think I found some clue to the problem:

Because the uno wifi R2 is a megaavr board and not an avr board, I went looking for the source code for Print.h. These apparently are not in the same location. From what I could find, there are at least two versions. I could not find the avr file (C:\Users\...\ArduinoData\packages\arduino\hardware\avr folder is blank)

However, the megaavr folder was not. I did find Print.h in:
(C:\Users\...\ArduinoData\packages\arduino\hardware\megaavr\1.8.6\cores\arduino\api)

While there, I noticed it defined the "Print" class within the "arduino" namespace.

```
namespace arduino {

class Print
{
```
I thought maybe there's another Print.h in a different namespace that is conflicting.

So, I edited the RGB LCD Shield header file to say:
`class Adafruit_RGBLCDShield : public arduino::Print {`
and it compiles for the Uno Wifi Rev 2. However, the arduino namespace appears to not be known to the avr boards, even if I use 
`using namespace arduino`
in the RGB_LCD header file.

I also checked and this behavior also happens with the liquid crystal library, upon which this is based, so maybe it's their issue to resolve?

dherrada, 2020-06-23:
Great. Thank you for opening this. I don't know C++ but I can try to get someone who does to take a look at this.

n-gineer, 2020-06-23:
I think I found a fix that doesn't break avr board use. New PR?

n-gineer, 2020-06-23:
Add to RGB_LCD.h
```
#ifdef ARDUINO_ARCH_MEGAAVR
using namespace arduino;
#endif

```

PR #21 (2020-06-24): Enable MEGA AVR Architecture compatibility
This enables MEGA AVR compatibility without breaking AVR compatibility as in previous pull request.

See Issue (closes #20) and previous pull request #17
closed 2020-06-24 by ladyada